adafruit/Adafruit_DAP #12: GCC 4.4 bitfield warnings

mzero, 2020-07-08:
The USERROW structures have non-portable uses of bitfields, as well as a large amount of code that essentially duplicates what the compiler should be doing with them.

Furthermore, GCC 4.4 now issues warnings about these structures like:
    note: offset of packed bit-field 'Adafruit_DAP_SAMx5::<unnamed union>::<unnamed struct>::BOD33_Level' has changed in GCC 4.4

These structures should be cleaned up to use more standard ways of defining machine registers, following the common patterns used in the CMSIS (which, oddly, doesn't have structure definitions for this particular register!)

I've begun work on such a fix, which you can find here:
[https://github.com/mzero/Adafruit_DAP/blob/userrow/Adafruit_DAP.h#L233](https://github.com/mzero/Adafruit_DAP/blob/userrow/Adafruit_DAP.h#L233)

Alas, this would be an API breaking change, though a relatively minor one, and I think worth it for consistency and correctness going forward.

Onno-Dirkzwager, 2021-04-20:
@mzero I ran into the same GCC compiler warnings...any chance you will make this into a PR?

hathach, 2021-04-24:
can you tell me which BSP you used for host. I don't see these warnings in samd and nrf.

Onno-Dirkzwager, 2021-04-25:
@hathach this is when compiling for an esp32 that flashes a samd21

...\lib\Adafruit_DAP\Adafruit_DAP.h:297:40: note: offset of packed bit-field 'Adafruit_DAP_SAMx5::<anonymous union>::<anonymous struct>::BOD33_Level' has changed in GCC 4.4
     struct __attribute__((__packed__)) {
                                        ^
...\lib\Adafruit_DAP\Adafruit_DAP.h:297:40: note: offset of packed bit-field 'Adafruit_DAP_SAMx5::<anonymous union>::<anonymous struct>::<anonymous>' has changed in GCC 4.4

hathach, 2021-04-27:
thanks for providing your setup. I have no idea why it complains, the GCC 4.4 is too old. I don't think we should worry about this if it doesn't affect the functionality. Yeah, it is not nice, but it is not urgent either. We can wait until somebody could make an PR to fix this.

JeffCalwood, 2022-11-05:
@mzero the proposed struct eliminate the compiler warning. BUT note that the size of the 'uint64_t BOD12_reserved_1 : 12;'  is incorrect! Size should be 8 bits, not 12!

PR #42 (2024-11-28): Fix samd nvm warnings
fix #12 , finally got some time to double check this with datasheet and hopefully fixes this warnings
closed 2024-11-28 by hathach