adafruit/Adafruit_DAP #13: stm32F103VET6 flash failed

zhizhuowu, 2020-07-28:
### hi, the example for stm32 flash from SD is not working on feather M0 adalogger with the following error message on serial printf:
Card initialized
Connecting...Adafruit Generic CMSIS-DAP Adapter 123456 v0.1 (S)
invalid response reading reg D (count = 1, value = 4)

that means the 
dap.dap_disconnect()  go through,
but 
dap.dap_connect() can't go through.

May I know whether there is a solution for that? i am using Arduino 1.8.3. does it have anything to do with the version?

ladyada, 2020-07-28:
it could be a wiring issue, could be speed, we have used it on teensy with success! its hard to debug, please use logic analyzer

closed 2020-07-28 by ladyada

zhizhuowu, 2020-07-28:
hi, Thank you!
May I know the setup you used to test it? I think it can be the speed. 
Thank you for your help.