adafruit/Adafruit_DAP #14: Compile Error

bfcooper, 2020-09-14:
I used your DAP program about 9 months ago, without any problems, and found it very useful.

However, when I try to use it now, I get a compile error.  I've tried it on two systems, but get the same result.  I've attached the verbose compiler log.

I hope you can get this straightened out.  It's a very useful program.
[DAP_errorLog.txt](https://github.com/adafruit/Adafruit_DAP/files/5221380/DAP_errorLog.txt)

ladyada, 2020-09-14:
hi, you can remove this line that is causing the error, we only use this library on platforms that have printf() support
Serial.printf("Unknown HWID = 0x%08X\n", hwid);

bfcooper, 2020-09-14:
Thanks for the prompt response....
That took care of the problem.

closed 2020-09-14 by bfcooper