adafruit/Adafruit_DAP #18: Adafruit_DAP_SAM::fuseRead/fuseWrite() buggy - erases reserved bits in the User Row

ksmith3036, 2021-01-05:
Adafruit_DAP_SAM::fuseRead() and Adafruit_DAP_SAM::fuseWrite() resets all reserved bits of the user Row to zero and because of this, the Voltage Regulator Internal BOD (BOD12) configuration is lost, since it is part of the User Row fuses. 

The SAMD21 datasheet says this about the Voltage Regulator Internal BOD (BOD12) configuration. These bits are written
in production and must not be changed.

The user row should be read in full, only the required bits changed, and then written back.

Before this is fixed, Adafruit_DAP_SAM::fuseWrite() should not be used.

hathach, 2021-01-19:
great analysis, would you mind making an PR to fix this one :wink: :wink:

ksmith3036, 2021-01-21:
Pull Request #16 fixes this, so this issue may now be closed.

closed 2021-01-21 by ksmith3036