adafruit/Adafruit_DAP #21: Enhancement Question: Add RP2040 support

SensorsIot, 2021-04-22:
Hi Lady Ada,
Would it be possible to add support for the RP2040?
(Adafruit_DAP_RP2040.cpp and Adafruit_DAP_RP2040.h)
Regards
Andreas Spiess

ladyada, 2021-04-22:
we would love to add more chips. please note we use UF2 to bootload for now, to add SWD please see if you can get the DAP commands so they can be added. note that it is a dual core, its not as simple as other chips

SensorsIot, 2021-04-22:
Thank you for your fast answer. I do not have the needed skills. That is why I was so happy to find this code. I will try to find somebody from my subscribers. We currently use your library to program a SAMD chip via SWD and it works fine. We will try to extend it to the STM32 family and maybe also to the RP2040 if we find the needed info...

hathach, 2021-04-24:
RP2040 make use of multi-drop SWJ sequence, therefore this library also need help to support it. There is some reference implementation in https://github.com/raspberrypi/picoprobe . Right now using host UF2 via USB is actually much faster than SWD (speed ~ 1-4 MHz, payload will be less).

SensorsIot, 2021-04-24:
@hathatch: We are searching for a standardized upload method for all ARM chips and would like to include the RP2040. This is why we tend to use SWD. It seems to be the minimal common denominator.
I will have a look at your link.

hathach, 2021-04-26:
For your purpose, swd is indeed the correct choice. As mentioned above, it is possible to get this repo to work with rp2040, it only needs
- DAP core need to be updated to support multi-drop
- Add rp2040 flashing specific sequence, probably involving invoking the rom API for flashing.

Either of those are supported by this repo, and we don't have plan to implement it as we prefer using msc uf2 for production flashing.

d-c-d, 2021-04-27:
> Either of those are supported by this repo, and we don't have plan to implement it as we prefer using msc uf2 for production flashing.

@hathach - just to confirm, by context I think that you meant "Neither" of those are supported, since there is no implementation plan ...

hathach, 2021-04-27:
> > Either of those are supported by this repo, and we don't have plan to implement it as we prefer using msc uf2 for production flashing.
> 
> @hathach - just to confirm, by context I think that you meant "Neither" of those are supported, since there is no implementation plan ...

AH right, just correct my comment

hathach, 2023-11-28:
closed due to lack of activities

closed 2023-11-28 by hathach