adafruit/Adafruit_DAP #25: ESP Examples PR questions

Onno-Dirkzwager, 2021-05-20:
Hi, 
I have made a few examples for when using an ESP as host:
- update an STM32 from SPIFFS
- update an SAMD21 from a browser

And would like to make a PR. But before doing so I  would like to ask some questions:
- Would you even be interested in these ESP examples?
- Right now these examples only work for the clients named above. Are you interested in examples for all clients? If so this would lead to multiple SPIFFS examples. And a pulldown with clients in the browser example.
- In a prev PR @hathach grouped the examples by mcu(client). Should I put my examples in the respective client directory or make a new one "ESP" named after the host?

ladyada, 2021-05-20:
hi! first up, does ESP work right now for DAP in general?
i think SPIFFS could cause a lot of issues - so there's going to be a lot of care to do a clean job of abstracting it! we'd be happy with a PR. please make a new folder just for ESP_SPIFFS

Onno-Dirkzwager, 2021-05-20:
Hi @ladyada,
It's a pleasant surprise to see such a busy person so involved in the issues & pr's. It's a honor!

**does ESP work right now for DAP in general?**
So far only been able to test on the SAMD21 & STM32. Outside of the bit field warnings mentioned [here](https://github.com/adafruit/Adafruit_DAP/issues/12) it works very well! I don't expect any different for the other clients. Both @SensorsIot and I have tested this from a server, browser and SPIFFS.

**SPIFFS could cause a lot of issues**
My initial tests and development when making this library compatible for use with the ESP's was done using SPIFFS as an alternative for the SD examples. Codewise they are near identical. So I don't see the issue. But I might be missing your point....

**Examples**
I'll make the folder as suggested and put the browser example in the SAMD21 folder.

ladyada, 2021-05-20:
ok best of luck! dont forget to also make sure CI passes :)

closed 2021-05-20 by Onno-Dirkzwager