adafruit/Adafruit_DAP #27: Question: Trying to get things to work with the Grand Central M4 board

cwbrandsdal, 2021-06-07:
This is not a issue to be honest, but I don't know where else to ask. I would like to use this library to flash a Grand Central board, but the only example I can find is for the Metro M4 with the bootlader as a header file. My question is how would I go about to make a similar header file, but for the Grand Central M4 Bootloader? Stuck between a rock and a hard place here, hehe.

The file in question: https://github.com/adafruit/Adafruit_DAP/blob/master/examples/samd51/samx5_flash_from_header/samd51_uf2.h

----
Christopher Brandsdal
Adafruit fandboy

cwbrandsdal, 2021-06-07:
Ok, so I was able to successfully create the array with some c# code (pasted it below if anyone wants it) and it flashes just perfect! But now I am very much stuck at next step. How would I combine the bootloader file with the actual firmware? This just flashes the bootloader, but I would like to also include my firmware. Just a simple blink sketch for now. :-)

```
void Main()
{
	HexDumper.HexDump(File.ReadAllBytes(@"C:\bootloader.bin")).Dump();
}

public static class HexDumper
{
	public static string HexDump(byte[] data)
	{
		var sb = new StringBuilder();

		for (int i = 0; i < data.Length; i += 12)
		{

			for (int j = i; j < i + 12; ++j)
			{
					sb.Append(" ");

				if (j < data.Length)
					sb.AppendFormat("0x{0:X2},", data[j]);
				else
					sb.Append("  ");
			}

			sb.AppendLine();
		}

		return sb.ToString();
	}
}
```

deladriere, 2021-06-10:
@cwbrandsdal Great Idea!
Today I use Bossac to read the chip containing the firmware uploaded with the  IDE (Arduino or Platformio)
As it contains the bootloader, I use it as a source for the Adafruit_DAP library.
Does it make sense ?
Merging both the firmware and the bootloader as suggested would be simpler.
I will keep watching this thread