adafruit/Adafruit_DAP #28: Example for MKR Boards: A hint for the serial monitor

CptHolzschnauz, 2021-07-28:
Thx for your amazing work!
Just a tiny little thing: For the Arduino MKRs when you have to choose which board, a little hint that "no line ending" has to be enabled to make the board choice work. With "Both NL & CR" it's not working.
Line 105 : Serial.print("Please select your board by entering the number. Please enable No Line Ending with 115200 Baud");

ladyada, 2021-07-28:
hi please submit a PR with a documentation update (we've never used MKR with this code)

CptHolzschnauz, 2021-07-28:
Lady Ada!
I'm ready to obey you! I did a PR but it's just a little hint without the need for a doc update, all the fantastic work was already done!
A big fan from Switzerland

ladyada, 2021-07-28:
thanx!!!

closed 2021-07-28 by ladyada