adafruit/Adafruit_DAP #30: Issues with programming NRF52840 and log

dazhbog, 2021-09-08:
Hi,

I am trying to program an NRF52840-xxD0 with the stock bin that was included in this repo's examples (40_BOOT.bin).

On a fresh reset, the chip replies and identifies correctly (see below). Right after the erase step, the programing step fails.

```
ID: 0x52840
Found Target: nRF52840_AAD0
Flash size: 1048576
Flash pages: 256
Correct, micro, reading SD..
Card initialized
Erasing... 
<<< 0x5 0x0 0x1 0x5 0x4 0xE5 0x1 0x40
>>> 0x5 0x1 0x1 0x0 0x0 0x0 0x0 0x0
<<< 0x5 0x0 0x1 0xD 0x2 0x0 0x0 0x0
>>> 0x5 0x1 0x1 0x0 0x0 0x0 0x0 0x0
[...]
>>> 0x5 0x1 0x1 0x0 0x0 0x0 0x0 0x0
<<< 0x5 0x0 0x1 0xF 0xC0 0x11 0xFF 0x1F
>>> 0x5 0x1 0x1 0x0 0x10 0x0 0x0 0x0
<<< 0x5 0x0 0x1 0x5 0x14 0x0 0x0 0x10
>>> 0x5 0x1 0x1 0x0 0x0 0x0 0x0 0x0
<<< 0x5 0x0 0x1 0xF 0xC0 0x11 0xFF 0x1F
>>> 0x5 0x1 0x1 0x0 0x1 0x0 0x0 0x0
 done.

Programming... 40_BOOT.bin
<<< 0x5 0x0 0x1 0x5 0x4 0xE5 0x1 0x40
>>> 0x5 0x0 0x3 0x0 0x0 0x0 0x0 0x0
invalid response writing to reg 5 (count = 0, value = 3)
```

For some reason, the micro replies with 0x05 (ok), 0x0 (should be 0x1) and 0x3 (3 is really weird as it cannot be in OK and WAIT state at the same time, should be 1 ideally). 

Im reading [this CMSIS reference](https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__Transfer.html), not sure if its correct
```
BYTE | BYTE **********| BYTE *************| WORD ********| WORD *********|
0x05 | Transfer Count | Transfer Response | TD_TimeStamp | Transfer Data |

Transfer Count: Number of transfers: 1 .. 255 that are executed.
Transfer Response: Contains information about last response from target Device.
Bit 2..0: ACK (Acknowledge) value:
1 = OK (for SWD protocol), OK or FAULT (for JTAG protocol),
2 = WAIT
4 = FAULT
7 = NO_ACK (no response from target)
Bit 3: 1 = Protocol Error (SWD)
Bit 4: 1 = Value Mismatch (Read Register with Value Match)
```

Another issue I have is that after a fail, I need to issue a reset with the J-link to make the micro behave again. Not sure if this is a separate issue. Are these kind of lock ups common?

I tried slowing down the clock in case its a timing issue. Code is the same as the example, with some printing here and there. 

If you see anything weird let me know, if not I will whip up the logic analyzer and check the tristate signals too.

Thanks
Marios

hathach, 2021-09-09:
can you provide all the hardware and software setup, board name, IDE, BSP version. and which sketch you are running for reproducing purpose.

dazhbog, 2021-09-09:
Sorry it was too late and forgot to give all the info.

Windows 10
Teensy 3.6 for the programmer 
Teensy Core: Teensyduino 1.54 (gcc version 5.4.1 20160919)
Arduino IDE is 1.8.9
Binary to load on device: binary in this repo or mine has the same effect
Custom PCB that works with 2 J-links with the same wiring (I will test the same setup with an adafruit board in the mean time)
DAP code is the same as this repo (same thing when using flash_feather52840.ino, SWD_standalone_core.ino)

Let me know if I missed anything!

dazhbog, 2021-11-16:
Just wanted to come back at this and say that I picked up this project again last night and I was able to make it flash. I am not sure why it wasn't working before could be wiring or something in the code.

My NRF is still not booting but the flashing seems to work ok so I shall play with it more and re-open if there is an issue.

closed 2021-11-16 by dazhbog

hathach, 2021-11-16:
Thanks for testing it again. Feel free to open an new issue if you find a bug