adafruit/Adafruit_DAP #31: Error when compiling flash_from_header

LordMurillo, 2021-12-15:
When trying to compile it displays the error:

`.arduinoIDE-unsaved20211115-3404-1t9rcw9.l34c\flash_from_header\2772cipy.h:21790:1: error: too many initializers for 'const uint8_t [0] {aka const unsigned char [0]}'
 };
 ^
\flash_from_header\2772cipy.h:21790:1: sorry, unimplemented: non-trivial designated initializers not supported
.arduinoIDE-unsaved20211115-3404-1t9rcw9.l34c\flash_from_header\2772cipy.h:21790:1: sorry, unimplemented: non-trivial designated initializers not supported
\flash_from_header\2772cipy.h:21790:1: sorry, unimplemented: non-trivial designated initializers not supported
\flash_from_header\2772cipy.h:21790:1: sorry, unimplemented: non-trivial designated initializers not supported
\flash_from_header\2772cipy.h:21790:1: sorry, unimplemented: non-trivial designated initializers not supported
\flash_from_header\2772cipy.h:21790:1: sorry, unimplemented: non-trivial designated initializers not supported
\flash_from_header\2772cipy.h:21790:1: sorry, unimplemented: non-trivial designated initializers not supported
\flash_from_header\2772cipy.h:21790:1: sorry, unimplemented: non-trivial designated initializers not supported`

Best regards

hathach, 2024-07-11:
which board, which example you are compiling ?

hathach, 2024-07-12:
this repo is mostly tested with samd, I don't think it support avr. Sorry we have no plan to support it either. PR is welcome though.

closed 2024-07-12 by hathach