adafruit/Adafruit_DAP #36: Compile failed with ESP32-C3 based boards

suraj022, 2023-01-29:
Hi

I've been trying to compile the code to work with seeed XIAO ESP32-C3 board. But it's showing an error as shown below.

I'm trying to load `flash_samd21_from_web_browser.ino` sketch. and getting the error.

c:\Users\Suraj\Documents\Arduino\libraries\Adafruit_DAP_library\Adafruit_DAP_SAMx5.cpp:152:31: warning: conversion from 'unsigned int' to 'uint16_t' {aka 'short unsigned int'} changes value from '4294967295' to '65535' [-Woverflow]
     _USER_ROW.bit.NVM_LOCKS = 0xffffffffu;
                               ^~~~~~~~~~~

Using library WiFi at version 2.0.0 in folder: C:\Users\Suraj\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\WiFi 
Using library WebServer at version 2.0.0 in folder: C:\Users\Suraj\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer 
Using library ESPmDNS at version 2.0.0 in folder: C:\Users\Suraj\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\ESPmDNS 
Using library Adafruit DAP library at version 1.8.0 in folder: C:\Users\Suraj\Documents\Arduino\libraries\Adafruit_DAP_library 
Using library FS at version 2.0.0 in folder: C:\Users\Suraj\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS 
exit status 1

Compilation error: exit status 1

any clues?

radio-satellites, 2024-06-02:
Just reading through this project... I don't believe this will work on an ESP32-C3, as this is for SAMD-based chips. The warning there complains about the overflow of the integer. 

Check out https://github.com/espressif/esp-serial-flasher for a similar project that works with the ESP32!

hathach, 2024-11-28:
should be fixed by https://github.com/adafruit/Adafruit_DAP/pull/42